pikvm/pikvm #1548: Incorporating certbot-dns-multi into kvmdcertbot

LeePorte, 2025-06-29:
**Is your feature request related to a problem? Please describe.**
Currently running DNS challenges for certificate renewals is not easy as many of the plugins are not installed or eays to install on the PiKVM platform

**Describe the solution you'd like**
Incorporating https://github.com/alexzorin/certbot-dns-multi into the kvmdcertbot package would enable wider DNS challenge usage as it supports 117 providers out of the box

**Describe alternatives you've considered**
I have considered using the manual challenge, which does work, but requires me to make interventions that are better handled by machines. Another alternative is suitable documentation on how to install this plugin on the PiKVM platform.

mdevaev, 2025-06-30:
I don't like the idea of adding such niche packages with a small number of users to the OS. You still install it from pip, isn't?

mdevaev, 2025-06-30:
Try `mkdir /root/foo` and `export GOTMPDIR=/root/foo` before building.

LeePorte, 2025-07-01:
I gave that a shot and still hit the no space left on device

```
[root@pikvm ~]# mkdir /root/foo
[root@pikvm ~]# export GOTMPDIR=/root/foo
[root@pikvm ~]# pipx install certbot-dns-multi
Fatal error from pip prevented installation. Full pip output in file:
    /root/.local/state/pipx/log/cmd_2025-07-01_19.39.44_pip_errors.log

pip failed to build package:
    certbot-dns-multi

Some possibly relevant errors from pip install:
    error: subprocess-exited-with-error
    github.com/hashicorp/go-multierror: unzip /tmp/tmp1gd5c7gf/pkg/mod/cache/download/github.com/hashicorp/go-multierror/@v/v1.1.1.zip: write /tmp/tmp1gd5c7gf/pkg/mod/github.com/hashicorp/go-multierror@v1.1.1/LICENSE: no space left on device
    subprocess.CalledProcessError: Command '('go', 'get', '-d')' returned non-zero exit status 1.
    ERROR: Failed to build installable wheels for some pyproject.toml based projects (certbot-dns-multi)

Error installing certbot-dns-multi.
[root@pikvm ~]# 
```

I also tried remounting `/var/lib/kvmd/msd/` as rw and creating a gotmpdir there and got the same result.

mdevaev, 2025-07-03:
Please try `TMPDIR` instead of `GOTMPDIR`.

LeePorte, 2025-07-03:
I switched to using `TMPDIR` and got the following

```
[root@pikvm ~]# rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
[root@pikvm ~]# mkdir /root/foo
[root@pikvm ~]# export TMPDIR=/root/foo
[root@pikvm ~]# pipx install certbot-dns-multi
Fatal error from pip prevented installation. Full pip output in file:
    /root/.local/state/pipx/log/cmd_2025-07-03_09.39.23_pip_errors.log

pip failed to build package:
    certbot-dns-multi

Some possibly relevant errors from pip install:
    error: subprocess-exited-with-error
    collect2: error: ld returned 1 exit status
    subprocess.CalledProcessError: Command '('go', 'build', '-buildmode=c-shared', '-o', '/root/foo/pip-install-6wmr9ms8/certbot-dns-multi_68bf2c12d9dc45d4922a8fc86311a4a4/build/lib.linux-armv7l-cpython-313/lego_bridge.cpython-313-arm-linux-gnueabihf.so')' returned non-zero exit status 1.
    ERROR: Failed to build installable wheels for some pyproject.toml based projects (certbot-dns-multi)

Error installing certbot-dns-multi.
[root@pikvm ~]#
```

An excerpt from  /root/.local/state/pipx/log/cmd_2025-07-03_09.39.23_pip_errors.log

```
      github.com/yandex-cloud/go-sdk/pkg/sdkerrors: mkdir /root/foo/go-build2264823109/b1388/: no space left on device
      github.com/hashicorp/errwrap: mkdir /root/foo/go-build2264823109/b1392/: no space left on device
      github.com/yandex-cloud/go-sdk/pkg/singleflight: mkdir /root/foo/go-build2264823109/b1393/: no space left on device
      golang.org/x/text/encoding/simplifiedchinese: mkdir /root/foo/go-build2264823109/b1186/: no space left on device
      github.com/go-acme/lego/v4/providers/dns/zoneee/internal: mkdir /root/foo/go-build2264823109/b1395/: no space left on device
      # github.com/goccy/go-json/internal/encoder/vm
      compile: writing output: write $WORK/b382/_pkg_.a: no space left on device
      go: failed to trim cache: write /root/.cache/go-build/trim.txt: no space left on device
```

Any other thoughts?

mdevaev, 2025-07-03:
Run `kvmd-helper-otgmsd-remount rw` and use `export TMPDIR=/var/lib/kvmd/msd/`. Don't forget to remove artifacts from previous tries.

mdevaev, 2025-07-04:
No, the problem is a lack of space in /tmp for building. You can try the shipped pip now with a working TMPDIR.

LeePorte, 2025-07-07:
I gave the shipped pip a try and this resulted in the following

```
[root@pikvm ~]# pip install certbot-dns-multi
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
[root@pikvm ~]# 
```

Is there any guidance on where to set up the python venv?

LeePorte, 2025-07-10:
It seems that installing with break system packages has broken kvmd-certbot

```
[root@pikvm ~]# rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
[root@pikvm ~]# kvmd-certbot renew
/usr/bin/kvmd-certbot: line 78:  6247 Bus error               sudo --preserve-env -u "$user" certbot renew "$@" --config-dir="$tmp/config" --work-dir="$tmp/work" --logs-dir="$tmp/logs" --deploy-hook="/usr/bin/touch '$tmp/updated'"
[root@pikvm ~]# kvmd-certbot help 
/usr/bin/kvmd-certbot: line 78:  6285 Bus error               sudo --preserve-env -u "$user" certbot "$@" "${cur_opts[@]}"
[root@pikvm ~]# 
```
reinstalling isn't a problem as it's only a home device, however it does mean that DNS challenges are a not viable currently

mdevaev, 2025-07-13:
Well, if this thing doesn't work on its own, I'm not sure how I can help you. Sorry.

LeePorte, 2025-07-13:
It does work when installing on Ubuntu using the certbot bundled pip.

intelfx, 2025-07-13:
> It does work when installing on Ubuntu using the certbot bundled pip.

@LeePorte OK, does it work on Ubuntu x86_64, or on Ubuntu on Raspberry Pi?

I reproduced the SIGBUS you quoted above. It appears to happen due to an unaligned memory access caused by the native code bridge in certbot-dns-multi. Unless it works for you on Ubuntu _on Pi_, there is nothing we can do to make it work on Pi-KVM. (Even if it by some miracle works on Ubuntu on Pi, it would indicate an ABI difference somewhere in Python or Go or libc, which means that you need to work with Arch Linux ARM to get it fixed.)

LeePorte, 2025-07-14:
I've only ever tried on x86_64. I have a spare pi I can test with Ubuntu to see if that works and then try on Arch Linux ARM.