pikvm/pikvm #1550: VNC via Jump Desktop performs poorly on iPhone over low bandwidth, unlike WebRTC interface; black & white / grayscale modes ineffective

plia7, 2025-07-01:
**Describe the bug**
When accessing PiKVM using Jump Desktop on an iPhone with a limited internet connection (~15 Mbps), the browser-based web interface works surprisingly well—even under constrained bandwidth—thanks to its WebRTC optimizations (e.g. JPEG quality 70%, max FPS 30, H.264 bitrate 5000 kbps, GOP 30, and absolute mouse mode). However, using the VNC server with Jump Desktop results in an extremely slow and unresponsive experience.

To compensate, I’ve tried using all available low-bandwidth settings in Jump Desktop such as:

Black and White

Grayscale

Low (8-bit) and High (16-bit) color

Despite selecting "Black and White" or "Grayscale", the image doesn't truly switch to those modes. Instead, it just appears as slightly reduced color depth. This suggests either the PiKVM VNC server isn't supporting those modes correctly or the client’s request isn't being honored, which severely limits performance.

Ultimately, I'd like to configure the PiKVM VNC server (e.g. kvmd-vnc) to behave more like the web UI, prioritizing performance with optimized compression and framerate handling. Is there a way to apply equivalent settings (e.g. JPEG quality, frame rate, bandwidth limits) on the VNC server? Or is the VNC path inherently limited due to protocol constraints?

This is especially important because the VNC client UI (Jump Desktop) is far more functional and usable than the PiKVM browser-based interface, which has significant limitations in mobile contexts. Being able to optimize the VNC experience would offer the best of both worlds: PiKVM’s excellent backend and Jump Desktop’s powerful frontend.

**Steps to Reproduce:**

1. Set up and run pikvm with default settings, including the web interface and built-in VNC server (`kvmd-vnc`).
2. Ensure the target machine (e.g. a server or computer connected via PiKVM) is powered on and accessible through both the web interface and VNC.
3. Connect your iPhone to a limited bandwidth connection (\~15 Mbps or lower).
4. From the iPhone:
   * Open **Safari or a WebRTC-compatible browser**, navigate to the PiKVM web UI, and verify that video streaming works reasonably well using default WebRTC settings (e.g. 70% JPEG quality, 30 FPS, H.264 bitrate 5000 kbps).
5. Now, open **Jump Desktop** on the same iPhone.
6. Add a new **VNC connection** to the PiKVM device and connect using various low-bandwidth profiles:

   * Try **Black and White**, **Grayscale**, **Low Color (8-bit)**, and other quality settings.
7. Observe that:
   * The VNC experience is extremely sluggish and far less responsive than the web interface.
   * Black and White and Grayscale modes **do not apply correctly**—the image retains colors or reduced color rather than full monochrome or greyscale.
8. Compare usability and responsiveness with the PiKVM web UI.

**Expected behavior**
VNC performance (via Jump Desktop) should be comparable to the PiKVM web interface, especially when using aggressive low-bandwidth settings.

"Black and White" and "Grayscale" modes should significantly reduce visual complexity and improve responsiveness over limited networks.

Reduced color profiles should result in faster screen updates, lower latency, and lower data usage.

Overall VNC experience should be usable and reasonably smooth, even under reduced bandwidth.

**Actual behavior**
VNC session via Jump Desktop remains very slow, unresponsive, and laggy even when using the most aggressive settings like "Black and White."

"Black and White" and "Grayscale" modes do not visibly take effect—the image appears as color-reduced, but not truly monochrome or grayscale.

Performance over the same connection is significantly worse than the web interface, despite the web UI delivering smooth video and responsive controls using WebRTC.

Due to these limitations, VNC is currently not usable in low-bandwidth mobile contexts, even though the Jump Desktop UI itself is much more user-friendly than the web interface.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
Windows 11
 - Browser [e.g. chrome, safari]
Google Chrome Version 130.0.6723.70 (Official Build) (64-bit) (cohort: Stable) 
 - Version [e.g. 22]
 - VNC client (if used)
 - iOS Jump Desktop app version 9.7.8

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 4.34-1
kvmd-fan 0.31-1
kvmd-platform-v3-hdmi-rpi4 4.34-1
kvmd-webterm 0.50-1
  - uStreamer version: `pacman -Q | grep ustreamer`
ustreamer 6.21-1
  - Linux kernel: `uname -a`
Linux pikvm 6.6.45-9-rpi #1 SMP Mon Dec 30 22:25:40 UTC 2024 armv7l GNU/Linux

plia7, 2025-07-02:
Given that Jump Desktop has discontinued active development on its VNC support and both it and bVNC rely on JPEG-based VNC streams, with no mobile VNC client supporting H.264 that I’m aware of, it really highlights the importance of the PiKVM web UI as the best long-term path for mobile access — since it already streams over H.264 efficiently.

I saw there’s ongoing work to improve the touch experience in [Issue #1075](https://github.com/pikvm/pikvm/issues/1075), and I appreciate the recent addition of **two-finger scrolling** (added in May) — that’s a great step forward. That said, the other points raised in that request — like **full-screen support**, **precise touch-to-cursor mapping**, **tap-to-click gestures**, and **pinch-to-zoom** — remain crucial for making PiKVM truly usable on phones and tablets, especially for remote workflows.

It’s just one more reason why mobile usability of the web interface matters so much — since it’s already the most bandwidth-efficient and codec-optimized option.

Also, do you know of any mobile VNC clients that currently support H.264 or similar hardware-accelerated codecs? That would be a game-changer for mobile users if available.

Thanks for all your work on this project and for considering mobile users in future enhancements.

mdevaev, 2025-07-03:
No, I don't know about such clients. We will continue to improve the interface, but this is a separate task.

closed 2025-07-03 by mdevaev