pikvm/pikvm #1552: Feature request: An example of capturing the audio/video webrtc stream with python

DIGist, 2025-07-04:
Feature request: An example of capturing the audio/video webrtc stream with python or bash.

This would help record useage and tutorials for documentation purposes.

Kaydron1000, 2025-07-12:
I am working on an application for something related to that. You should be able to pipe output of api/media/ws to ffmpeg and get a video. this should all be able to be done in powershell. 

Additionally something exists. You record on PiKVM then pull off of SD card currently. Please see:
https://docs.pikvm.org/video/?h=record#video-recording

DIGist, 2025-08-21:
@Kaydron1000 Interested, I'd love to see what you have.
I have a python script that is semi working, it authenticates and records the Video and Audio, but I'm missing something cause I get some weird aiortc errors and some of the frames drop, oh and it seems to only record the top-left 640x470 quadrant, I think that's related to some ffmpeg setting.

As for recording video stream on the device, that works great, but I still need audio and I don't know of a way to capture the audio into the mp4 file.

DIGist, 2025-10-28:
Thanks for all the work!

I made a prototype script working to record the WEBRTC with video and audio, video waits a number of seconds, I assume waiting for a keyframe or something, audio works for the duration. it would almost be all workable, there is some bug either in aiortc or it's interface to ffmpeg that is dropping alot of frames. (I have a ticket open with aiortc about the ffmpeg frame issue).

```
#!/bin/python
#tested with python 3.12.9 and aiortc=1.11
# Required PyPI packages:
#   aiortc
#   websockets
#   cryptography (dependency of aiortc)
#   av (for media encoding)

import argparse
import asyncio
import json
import logging
import os
import secrets
import ssl
from contextlib import asynccontextmanager
from aiortc import RTCPeerConnection, RTCSessionDescription, RTCIceCandidate
from aiortc.contrib.media import MediaRecorder
import websockets

logging.basicConfig(level=logging.INFO)
logger = logging.getLogger("kvm_recorder")

class KVMRecorderError(Exception):
    """Custom exception for KVM recording errors."""
    pass

@asynccontextmanager
async def _kvm_websocket_connect(hostname):
    """Context manager for authenticated KVM WebSocket connections."""
    username = os.environ.get("piUSERNAME")
    password = os.environ.get("piPASSWORD")

    if not username or not password:
        logger.error("piUSERNAME and piPASSWORD environment variables must be set.")
        raise KVMRecorderError("Missing KVM credentials in environment variables.")

    uri = f"wss://{hostname}/api/ws?stream=0"
    headers = {"X-KVMD-User": username, "X-KVMD-Passwd": password}
    ssl_context = ssl._create_unverified_context()

    ws = None
    try:
        ws = await websockets.connect(uri, additional_headers=headers, ssl=ssl_context)
        yield ws
    except Exception as e:
        logger.error(f"Failed to connect to WebSocket: {e}")
        raise KVMRecorderError(f"WebSocket connection failed: {e}")
    finally:
        if ws:
            await ws.close()

async def janus_send(ws, message, timeout=10):
    """Send a message to Janus and wait for the matching response transaction."""
    txn = secrets.token_hex(8)
    message['transaction'] = txn
    
    try:
        await ws.send(json.dumps(message))
        
        # Use asyncio.wait_for for timeout and collect messages in order
        pending_messages = []
        while True:
            try:
                raw = await asyncio.wait_for(ws.recv(), timeout=timeout)
[... 246 lines cut ...]
        return 1
    except Exception as e:
        logger.error(f"Unexpected error: {e}")
        return 1
    
    return 0

if __name__ == '__main__':
    parser = argparse.ArgumentParser(description="Record KVM WebRTC stream via Janus")
    parser.add_argument("-o", "--output", default="pikvmrecord.mp4",
                        help="Output file name")
    parser.add_argument("-H", "--host", required=True,
                        help="Hostname or IP of the KVM server")
    parser.add_argument("-r", "--retries", type=int, default=3,
                        help="Maximum number of retry attempts")
    args = parser.parse_args()

    try:
        exit_code = asyncio.run(main(args.host, args.output, args.retries))
        exit(exit_code)
    except KeyboardInterrupt:
        logger.info("Interrupted by user")
        exit(130)

```

mdevaev, 2025-12-11:
No, there is no way to do this now because audio is captured directly by janus plugin and it handles an audio itself inside. In case of video, we have ability to redirect streams. That's why we need to rewrite the entire plugin to split audio functions.

DIGist, 2025-12-11:
So interesting enough, In trying to find if there was better way to capture the audio and video, Janus Server Gateway claims it supports audio/video WEBRTC recording out of the box? https://janus.conf.meetecho.com/docs/recordings.html

I'm just scratching the surface on what that might look like to try to access, but if you have any thoughts in the mean time, I'd love to hear them.

mdevaev, 2025-12-11:
It still require coding. I think aiortc is a better way now.