pikvm/pikvm #1569: No Image after PowerOn with USB connected on Dell Optiplex/Pro Micro 7020 Desktops

freaky-m0, 2025-08-04:
**Describe the bug**
I use PiKVM 4+ and PiKVM Switch with Dell Optiplex/Pro Micro 7020 Desktop Devices. When I exit the BIOS or PowerOn the device I get no image (in the browser and on the passthrough hdmi) and weird behaviour (bootloop, no date / BIOS Settings after next Boot without USB) while the USB uplink cable to PiKVM is connected. When I disconnect the USB cable everything works as expected.

I already tried different EDIDs, USB Ports, BIOS Settings, without USB Passthrough, without external Monitor etc.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Restart Dell Optiplex/Pro Micro 7020 Desktop Computer'
2. Enter BIOS with F2
3. Exit BIOS
4. Device will don't show Image / not boot correctly

**Expected behavior**
Image and USB working after PowerOn / BIOS exit

**Screenshots**
This setting also helps instead of disconnecting the USB cable physically:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Firefox

**PiKVM info:**
  - Raspberry Pi board version (it was a complete device I guess with RPi4? no DIY)
  - PiKVM platform V4+ with PiKVM Switch
  - Video capture type: ? Should be the default
  - KVMD version:
          kvmd 4.92-1
          kvmd-fan 0.32-1
          kvmd-platform-v4plus-hdmi-rpi4 4.92-1
          kvmd-webterm 0.50-1
  - uStreamer version: ustreamer 6.40-1
  - Linux kernel: Linux pikvm 6.6.45-13-rpi #1 SMP PREEMPT Tue May 27 17:29:07 UTC 2025 aarch64 GNU/Linux

freaky-m0, 2025-08-06:
Disabling mass storage with kvmd-otgconf helps. I can turn off and on the device and exit the BIOS without any problems.

Can I help you further with any additional information?

mdevaev, 2025-08-06:
This is a defect in the DELL BIOS. Or rather, a defect in the framework that they and HP use for some chipsets. We are exploring possible workarounds for this.

mdevaev, 2025-09-16:
Try this:
```yaml
otg:
    devices:
        msd:
            start: false
```

mdevaev, 2025-11-09:
I'll use this issue to track our research on this topic. If we find a generalized solution, it will be a celebration.

mdevaev, 2025-11-11:
@prawilny It does almost the same thing from the host point of view. Probably you missed something in the config.