pikvm/pikvm #1583: Problem uploading large files to USB drive

softfoot, 2025-09-14:
**Describe the bug**
I have a V2 and have a samba share and usb drive in "fstab" setup to be available in the "disk" menu.
I can upload large files (2Gb) to "internal" and the "samba share" without problems.  But, I unable to 
upload anything bigger than ~400Mb to he USB drive - it times out at 1% and I get an error message
"cant upload file".

**To Reproduce**
Format a USB stick as exfat and put the following in fstab -
   //192.168.2.4/MainArchive/PiKVM-images   /var/lib/kvmd/msd/MainArchive cifs guest,_netdev,nofail 0 0
   /dev/sda1 /var/lib/kvmd/msd/usb exfat auto,nofail,rw,umask=0000 0 0
and reboot.
Try to upload a file that is say 2Gb

**Expected behavior**
Should upload

**Desktop (please complete the following information):**
 - OS: Win7 or Win10
 - Browser Firefox]
 - Version [e.g. 24

**PiKVM info:**

Base: Raspberry Pi 4 Model B Rev 1.2 Platform: v2-hdmiusb-rpi4 Serial: 10000000F5F17D6E KVMD: 4.95 Streamer: 6.40 (ustreamer)

    WITH_GPIO: Yes
    WITH_JANUS: Yes
    WITH_PDEATHSIG: Yes
    WITH_PTHREAD_NP: Yes
    WITH_PYTHON: Yes
    WITH_SETPROCTITLE: Yes
    WITH_SYSTEMD: Yes
    WITH_V4P: Yes

Linux kernel:

    Machine: aarch64
    Release: 6.6.45-13-rpi
    Version: #1 SMP PREEMPT Tue May 27 17:29:07 UTC 2025

**Additional context**
Add any other context about the problem here.

mdevaev, 2025-11-09:
> A Kingston Traveler 128G USB 3 drives work but ONLY in USB 3 ports.
> A Kingston Traveler 259G USB 3 drive fails in either USB port type.

Could you make some speed test on windows?

> As an aside I notied that when I changed the drive the "drive" menu did not detect the new usb drive. 

This happens because as soon as you have mounted the drive, the OS continues to assume that it is there. We do not have automatic media disconnection and connection. Reset will re-scan the existing file structure.

softfoot, 2025-11-23:
I sent this on the 16th 👎 

"In a USB 3 slot under WIN10 ... 256 Gb drive wr=16Gb/s rd=95Gb/s 128 Gb drive wr=34  rd=95"